spleenware/ripple #1: OLED on Heltec V2 board

matthewrwright, 2019-09-18:
Heltec V1 and V2 boards have different pinouts for the OLED display. 

Is it possible to have a version compiled for the V2? 

Changes required should be 

#include <U8x8lib.h>

//For Heltec Wifi LoRa 32, TTGO LoRa and TTGO LoRa32 V1 use:
U8X8_SSD1306_128X64_NONAME_HW_I2C display(/*rst*/ 16, /*scl*/ 15, /*sda*/ 4);

//For TTGO LoRa32 V2 use:
U8X8_SSD1306_128X64_NONAME_HW_I2C display(/*rst*/ U8X8_PIN_NONE);

Happy to test and report back on new firmware.

spleenware, 2019-09-19:
OK, I have prepared builds for the V2 board. See the readme. Hope it works! :-)

matthewrwright, 2019-09-19:
Unfortunately not - but it looks like on this particular board there are some issues I can't quite find

Using SW_I2C works, using HW_I2C on the same pins doesn't - And confusingly the constructor for the two calls is different!

U8X8_SSD1306_128X64_NONAME_SW_I2C display(/* clock=*/ 15, /* data=*/ 4, /* reset=*/ 16);

The above line works for me (tested with my board)

//U8X8_SSD1306_128X64_NONAME_HW_I2C display(/*reset*/ 16, /*clock*/ 15, /*data*/ 4);
//fails
//U8X8_SSD1306_128X64_NONAME_HW_I2C display(/*reset*/ 16);
//fails

Those should also work, however they're not at all!

Is there a chance you could compile up a SW_I2C version so I can give it a try ?

spleenware, 2019-09-19:
So, does the board connect with the Android app OK? (just curious)

spleenware, 2019-09-19:
I am using this Arduino library: https://github.com/ThingPulse/esp8266-oled-ssd1306
And for your board, have compiled with this constructor: 
   SSD1306  display(0x3c, 21, 22);   // V2
 
Can you verify if this library CAN work with this board?

matthewrwright, 2019-09-19:
Connects fine with the android app, not a problem via Bluetooth - I don't have an appropriate OTG cable handy to try the other way. 

Will test that library shortly for you.

matthewrwright, 2019-09-19:
I've had a good hour with that library and I can't make it work with my board at all unfortunately.

spleenware, 2019-09-20:
Can you paste a URL to the library you know that works with your board?

matthewrwright, 2019-09-20:
It's this one

https://github.com/olikraus/u8g2

Might be handy for the future as it supports other displays too

spleenware, 2019-09-23:
OK, I've rebuilt the heltec V2 binaries (USB and Bluetooth Classic) with the U8G2 library.
Have used this constructor (hope it's the right one):
U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R2, /* reset=*/ U8X8_PIN_NONE);

matthewrwright, 2019-09-23:
Cheers - Doesn't seem happy. Can we try this? 

U8X8_SSD1306_128X64_NONAME_SW_I2C display(/* clock=/ 15, / data=/ 4, / reset=*/ 16);

I've tested that with a blank sketch for a "Hello World" and it works - can't get HW_I2C working, even with the same pins, I'm guessing it's a bug in the board.

spleenware, 2019-09-23:
Have rebuilt, using the following constructor. (Note: I haven't read up on the 'U8X8...' stuff, which is just a text-only mode).

U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R2, /* clock=*/ 15, /* data=*/ 4, /* reset=*/ 16);

spleenware, 2020-11-26:
I have verified that the latest Heltec builds have the OLED working now.

closed 2020-11-26 by spleenware