spleenware/ripple #6: Not working on TTGO ESP32 LORA

valerio-vaccaro, 2019-10-18:
After flashing on a TTGO ESP32 LORA (using command provided in the readme) I have no feedback (oled black), serial port reply just with bootloader messages (at baud rate 115200)
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
```

valerio-vaccaro, 2019-10-19:
Continue reboot also on `Heltech 433` board using firmware `Heltech ESP32 LORA V2 - 433MHz, for USB-OTH with handset`

```
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
```

spleenware, 2019-10-20:
I'm not sure how to diagnose this. So, did the firmware flashing succeed? If it did, but the board just seems to not be responding, try opening the serial monitor and enter "i0,0" newline(without the ")
It should respond with "I:126,1"
Oh, serial monitor baud at 9600

valerio-vaccaro, 2019-10-27:
Firmware correctly flashed, just no answer for 868 board (9600, sending an i0,0 string) and continuouse reboot for 433

spleenware, 2019-10-27:
I could try ordering one of the exact boards you have and try myself. Otherwise, it's probably near impossible to diagnose what could be happening. What is the buy link?

guywire, 2019-11-28:
I'm getting the same message from my TTGO ESP32 LORA. I couldn't get the command line to upload as written so I used the ESP32 Download Tool.

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8

guywire, 2019-11-29:
I've tried a few extra things, thought I made progress, but pretty much the same results.
Installed Python 2.7.1.17.a - it appears you need python to run command line esptool
Grabbed earlier 1.02 version of arduino-esp32 which appears to match bin files you're using from here https://github.com/espressif/arduino-esp32/releases/tag/1.0.2-rc2 - notice there's and rc1 and rc2, tried both
modified command line so it would run:
D:\User\Downloads\esp32-1.0.2-rc2\tools>esptool.py --chip esp32 --port COM10 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 partitions\boot_app0.bin 0x1000 sdk\bin\bootloader_qio_40m.bin 0x10000 Ripple\Ripple-915-USB.heltec_wifi_lora_32.bin 0x8000 Ripple\Ripple-915-USB.partitions.bin
esptool.py v2.6
Serial port COM10
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 84:0d:8e:05:a7:24
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4681.1 kbit/s)...
Hash of data verified.
Compressed 16848 bytes to 10926...
Wrote 16848 bytes (10926 compressed) at 0x00001000 in 0.1 seconds (effective 983.8 kbit/s)...
Hash of data verified.
Compressed 246384 bytes to 126102...
Wrote 246384 bytes (126102 compressed) at 0x00010000 in 2.6 seconds (effective 772.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 1890.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

D:\User\Downloads\esp32-1.0.2-rc2\tools>

Now getting this:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8

This is a link to the board, which is V1 of the TTGO ESP43 LoRa 
https://www.banggood.com/2Pcs-LILYGO-TTGO-LORA32-915Mhz-ESP32-LoRa-OLED-0_96-Inch-Blue-Display-p-1239769.html?rmmds=myorder&cur_warehouse=USA

spleenware, 2019-11-29:
That first invocation looks like it was successful. When it says
"Leaving... Hard resetting via RTS pin..." usually indicates success.

Have you tried opening the serial monitor in Arduino IDE, set baud rate to
9600, also the NEWLINE option, and enter 'i0,0' (without the quotes) and
press enter. If you get a response starting with 'I:" the all is good.

For the gps tracker and sensor firmware, there is a different serial
protocol,and to check you just enter 'q' in serial monitor. Response should
start with 'Q:'

guywire, 2019-11-30:
Aha, that's not how I thought baud worked. With the TTGO ESP32 LORA - 915MHz, for USB-OTG with handset build I get a response to "i0,0" at 9600 baud, and it works with the app and is able to transmit and receive ascii text! 

I recommend having the oled screen display something to show it's working when it boots successfully, and maybe to show it's connected to the phone.

Unfortunately I can't get any response using the Bluetooth Classic or Bluetooth LE builds.

This is the BLE install, which looks like it loads successfullyl:
D:\User\Downloads\esp32-1.0.2-rc2\tools>esptool.py --chip esp32 --port COM10 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 partitions\boot_app0.bin 0x1000 sdk\bin\bootloader_qio_40m.bin 0x10000 Ripple\Ripple-915-BLE.heltec_wifi_lora_32.bin 0x8000 Ripple\Ripple-915-BLE.partitions.bin
esptool.py v2.6
Serial port COM10
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 84:0d:8e:05:a7:24
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5461.4 kbit/s)...
Hash of data verified.
Compressed 16848 bytes to 10926...
Wrote 16848 bytes (10926 compressed) at 0x00001000 in 0.1 seconds (effective 976.7 kbit/s)...
Hash of data verified.
Compressed 1219152 bytes to 682674...
Wrote 1219152 bytes (682674 compressed) at 0x00010000 in 14.5 seconds (effective 674.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 1755.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Serial Monitor at 115200 baud:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8

Serial Monitor at 9600 baud shows no response to "i0,0" or "q"
shows this upon reboot ⸮`#Ee⸮⸮S⸮⸮1o⸮

It does show up in the app preferences  - choose device as a Ripple Service, let's you select it, but doesn't appear to connect, the bluetooth icon doesn't light up. It's mostly the same for both BLE and Classic. (The classic looks like this at 9600 baud !⸮d9D⸮e⸮⸮⸮ 6⸮⸮⸮0)

spleenware, 2019-11-30:
thanks for the suggestion. Yes, it would make sense to display something on the OLED on bootup.
BLE build is still pretty flaky. Classic seems to work OK. You just have to do the Pair step yourself on the handset, then select the device in the Setup screen in the app.
I haven't tried using the Serial Monitor with the Classic build, ie. using laptop's bluetooth etc

guywire, 2019-11-30:
Ah, my bad, I missed the bluetooth classic instructions to pair it in bluetooth settings. It works.
Yes BLE also sort of works too, but I see it causes errors in the text and times out.
So it looks like this board works well for bluetooth classic and OTG if you do do it all right.

I noticed you can send emoji but they just get received as ? boxes. Just since the power talkies can send emoji that would be cool if it worked.
I'll have to get some more boards to try the mesh feature.
Thank you for helping me troubleshoot.

guywire, 2019-11-30:
And of course today after making both my devices bluetooth classic emoji seem to be working too.

Nolan204, 2020-01-23:
I am having the same issue as Guywire but I can't even get Classic to work, Have not tried USB tethered, The new firmware you created loads fine shows the splash screen, so 0.0V but shows it is connected but I cannot send a msg when I try the screen does light up but the msg fails on retires. the serial monitor sends gibberish at 9600 baud, 115200 shows something but i0,0 does nothing. Any idea or these boards are just not very good?

valerio-vaccaro, 2020-01-25:
TTGO/Heltec Lora 868 working via USB/BLE with last firmware.
433 still have some issues