spleenware/ripple #9: Ripple-433-Bluetooth.heltec_v2.bin

luxonn, 2019-12-13:
Android program ends with an error while receiving geo coordinates.

spleenware, 2019-12-20:
Hi, OK I have a crash report for that now from Play. I'll get a fix out for that this weekend.

luxonn, 2019-12-20:
When more than More than two pagers, one of the three is not in communication. I still don’t understand if relaying works. Do pagers form a network? 
Which means: "Unable to update friends: TimeoutException: null" Such a message often happens, especially when there are three pagers.

spleenware, 2019-12-21:
> When more than More than two pagers, one of the three is not in communication. I still don’t understand if relaying works. Do pagers form a network?
> Which means: "Unable to update friends: TimeoutException: null" Such a message often happens, especially when there are three pagers.

I'm still not sure what causes that error, but I have only reproduced it a couple of times. Most the time, simply unplugging the USB cable, and plugging in again clears everything up.

Are you using one of the Bluetooth builds, or USB OTG?

spleenware, 2019-12-21:
OK, I have published ver 1.2.1 of the Ripple Messenger app on Google Play. It should contain the geo location bug fix.

luxonn, 2019-12-27:
I use only bluetooth. It is convenient and not all have the OTG cables

closed 2020-01-14 by luxonn