spleenware/ripple #10: BSFRance LoRa32u4 II

lburrowes, 2019-12-20:
Had a old board laying around and wanted to try your code on this board.  
Atmega32u4  and SX1276

ichrised, 2020-04-11:
Did it work? I bought the adafruit feather lora m0 and I want to make repeater nodes but the adafruit ones are quite cost prohibitive

lburrowes, 2020-04-11:
I will circle back on this.  Since ripple was so restricted with code on how it works I went to check out meshtastic and disaster radio builds.

spleenware, 2020-04-11:
Hi,
I bought a TTGO T-Deer to have a crack at this (it has a 32u4 architecture), but it simply doesn't have enough RAM. I have, however, had success with the Seeed Xiao, which has Cortex M0, which is much cheaper than the Feather. It doesn't have LoRa radio, but I got a cheap RFM95W module on ebay. I'll be uploading the firmware image soon (for the Xiao)