spleenware/ripple #11: heltec 433 overlap @ 0xe000

allegiant1, 2019-12-28:
I am trying to flash the bin file and i have tried on windows and mac and linux and have had same error all 3 machines trying to flash the bin with esptool.  error says 

esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0xe000 for file: /home/demo/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/partitions/boot_app0.bin

Its probably user error but I have searched for solutions and cannot find any.
any help would be greatly appreciated.

spleenware, 2019-12-28:
Try opening one of the simple Arduino sketches, in examples for instance, select the appropriate board type and port etc, then upload to the board. (Beforehand, go to the Arduino Preferences, and check/tick the "Show Verbose output during:   [ ] Upload"
In the log area at the bottom, during the build/upload, you should see the full command-line command that was used to upload to that board. (will prob have ... esptool ... in it). The last two '.bin' files should be the application that was just compiled. Try this same command, but replace the last two '.bin' file names with the Ripple bin file names

allegiant1, 2019-12-28:
uploaded sketch successfully.... but get same overlap error when copying the output and pasting for to the command line and replacing last 2 files with Ripple files.

allegiant1, 2019-12-28:
I get the same overlap error with TTGO board and with Heltec boards both 433 and 915 bluetooth or USB.  I have tried all different boards and get same error on windows/mac/linux. Something I am doing wrong?

spleenware, 2019-12-28:
Could you copy/paste the full command-line command you are using here?

allegiant1, 2019-12-29:
here is one of my attempts.
thank you for your help.

python /home/demo/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /home/demo/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/partitions/boot_app0.bin 0x1000 /home/demo/.arduino15/packages/esp32/hardware/esp32/1.0.3/tools/sdk/bin/bootloader_qio_80m.bin 0x10000 ~/Desktop/LORA/Ripple-915-Bluetooth.heltec_v2_wifi_lora_32.bin 0x8000 ~/Desktop/LORA/Ripple-915-Bluetooth.heltech_v2.partitions.bin

allegiant1, 2020-01-03:
fresh install of windows7 and arduino ide and esp32 espressif tools yields the same overlap error.

allegiant1, 2020-01-04:
Fresh install of mint Linux and windows 8 .. Arduino idea 8.10 espressif esptool 2.6.1 and 1.0.4 bootloader. Same overlap at 0xe000 addressing error... Any help?

allegiant1, 2020-01-06:
Could it be a different version of esptool?

spleenware, 2020-01-07:
I've been Googling too, and I really don't know what to suggest :-(