spleenware/ripple #12: Which Board to choose?

lyrics1988123, 2020-01-02:
i want to test your project as i think its a great idea, but im not sure which hardware would be the best for my case? (i need 868mhz 100mw at least)
i really want to have the bluetooth connection with at least 30m range instead of an OTG Cable (for nice real-word positioning).. any advice which one to choose? 

thanks :)

spleenware, 2020-01-03:
Most people seem to have most success with the Heltech boards. I don't have
any direct links, but try to find one that offers an IPEX antenna
connector. Makes it easy to connect a *proper* antenna, not the BS ones
they seem to include. I'd recommend this one:
https://www.banggood.com/Original-FrSky-868MHz-Dipole-T-Antenna-IPEXI-for-R9-Slim-Slim-Receiver-LBT-Version-RC-Drone-p-1380277.html?rmmds=search&cur_warehouse=GWTR

lyrics1988123, 2020-01-03:
so this one should work? wich firmware would i need then?

https://www.amazon.de/MakerHawk-ESP32-Entwicklungsboard-OLED-Display-64M-Bit-Flash-intelligente/dp/B07ML6528K/ref=sr_1_4?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=esp32%2Bheltec&qid=1578018790&s=ce-de&sr=1-4&th=1

spleenware, 2020-01-03:
Yes, that should be fine. I'm not sure if this is a V1 or V2 Heltech, but I
have firmware for both.

lyrics1988123, 2020-01-11:
its a SX1276 lora chip..
but still not sure which firmware i would need for bluetooth with this board :/

spleenware, 2020-01-11:
I'm soon to release a new Bluetooth build for these boards (BLE, that is).

lyrics1988123, 2020-01-11:
ah okey, so the existing ones are not the correct ones..