spleenware/ripple #13: Commander App

dwhacks, 2020-01-12:
So I bought the commander app, and got two ttgo boards. I don`t have two android this instant to test, but I believe I got the firmwares loaded up (serial monitor replies "I:126,1" when I send "i0,0"

Bluetooth seems to connect to phone, and in both the messenger and commander app the bluetooth symbol at the top lights up when plugged in the OTG cable.

Thing is, commander app says "no devices" and I can't see anywhere to add any settings or anything. I'm pretty new to OTG and not really sure if I'm doing something wrong.

Also both OLED screen seem to display mystery (to me) info. 
R:-157 F:0 B:0 T:0 and a few more lines, with some counting going on. The USB otg one has a (000000) and the bluetooth (100000) as the only differences.

spleenware, 2020-01-12:
Hi,

Thanks for the purchase! The documentation is still pretty sparse for the Commander app, so my apologies.
The app will only setup (and monitor) boards using the USB OTG connection. There is no Bluetooth (classic or BLE) support here. Bluetooth is only for the Messenger app.

For setting up new nodes, ie. repeaters, sensors or GPS trackers, you need to launch with the 'Device Provision' launcher icon. (there are TWO launcher icons) Here you will see a list of defined nodes/devices. In the actionbar there are menus 'NEW' for adding a new sensor or GPS tracker node. And there is a 'SCAN' menu for importing 'pager' nodes, ie. people you want to chat with (just like in the Messenger app).

The OLED screens display some low level diagnostics, which probably shouldn't have been in the firmware releases. :-( I will be releasing new firmware binaries soon with better OLED display.

dwhacks, 2020-01-12:
Great! thats a start! I've found the other launcher and created a "repeater" (useless as I only have two devices right now, but 2 more are coming).

now in the regular comander app it shows my repeater, and when I tap on it it says:
Type: Repeater
RSSI Timed Out
Battery ....

It has a lipo connected so I assume it should tell me something. 

Either way, keep up the great work. I hope some of this will help you get some documentation out at some point.

jobionekabnoi, 2020-01-13:
I've experienced the exact same issues w/ the Commander app with the following LoRa nodes: 
-TTGO ESP32 Lora with OLED
-Heltech V2 board
-TTGO Beam

Also, is there any way to deletes nodes from the app?

Definitely agree with dwhacks, keep up the great work and looking forward to the updates!

spleenware, 2020-01-13:
Regarding the Timed Out issue:

This usually happens if the encryption keys haven't been setup (or the other device is actually offline).

At present you need to make the Home node have ID 1. This is the ID of the node you attach to the Commander handset itself. 
The repeater/sensor/tracker nodes will persist the encryption shared secret (in EEPROM) so that it can communicate back and forth with node 1. (even after a reboot)
This is done during the 'Program' step. In the Device Provision list, there is a small 'computer chip' icon for each node. Tap on these to enter the Program screen. (Have you done this step?) Anyway, when in that screen, plug in the repeater/sensor/tracker node via USB, then click the 'Program' button.
This will flash the device with the ID and exchange keys

spleenware, 2020-01-14:
BTW, I am fleshing out a user guide here in the Wiki: https://github.com/spleenware/ripple/wiki/Commander-app-user-guide