spleenware/ripple #14: how to flash it?

lyrics1988123, 2020-02-04:
guys im just too stupid. how can i use the commandline code?
i need the following but when i try to run it, my cmd tells me, that the command "~" is wrong or not existant (windows).. never flashed an esp32 before.. please help >.<

~/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/partitions/boot_app0.bin 0x1000 ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 Ripple-BLE.TTGO-V1.bin 0x8000 Ripple-BLE.TTGO-V1.partitions.bin

spleenware, 2020-02-04:
Hi,
The '~' just means the user's home directory. For Windows you need to find where the esptool.exe is (which I think you have figured out). With the error above, change the '/' to a '\' before esptool

spleenware, 2020-02-04:
ah, this editor masked the 'backslash' I typed in. Change the forward slash to a backslash before esptool.exe

lyrics1988123, 2020-02-04:
hey, thanks for the very fast reply!
it was a typo.. oh lol
but i still get the same issue:

```
C:\Users\waiss\Downloads\lora>C:\Users\waiss\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1\esptool.exe --chip esp32 --port COM4 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\waiss\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\partitions\boot_app0.bin 0x1000 C:\Users\waiss\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\sdk\bin\bootloader_dio_80m.bin 0x10000 Ripple-BLE.Heltech-v2.bin 0x8000 Ripple-BLE.Heltech-v2.partitions.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0xe000 for file: C:\Users\waiss\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\partitions\boot_app0.bin

C:\Users\waiss\Downloads\lora>
```

lyrics1988123, 2020-02-04:
tried it with my 2nd board, same issue.

dwhacks, 2020-02-04:
Try using the ESP loader programme directly available from esp. I found it to be the easiest way on windows.

lyrics1988123, 2020-02-05:
tried that.. and it looks like its flashing but i still have the standard programm running on it (a wifi test sketch).. =(

lyrics1988123, 2020-02-05:
any idea? really wanted to try it out =(

spleenware, 2020-02-06:
> any idea? really wanted to try it out =(

I have done a build on a Windows machine, for the Heltec V2 target, and captured the install commands and BIN files. Hopefully these work on your machine. Download this zip: https://drive.google.com/open?id=1dh9NOvUL0unOFCqyFuoMIfoMbTZG_yHT

spleenware, 2020-02-06:
> any idea? really wanted to try it out =(

Also, I made two builds, the USB one and Bluetooth Classic.

lyrics1988123, 2020-02-06:
thought this board has BL "LE" like you said in the other thread? x)
but thank you very much! will try it in the evening!
will let you know

spleenware, 2020-02-06:
Yes, it has BLE and Classic, but the Classic has been more reliable in my testing. I 'thought' that I had the BLE stuff working well, but I'm still getting problems with it :-(

lyrics1988123, 2020-02-06:
flashing did work (but the displays shows nothing if it should..)

i was able to connect to it with ripple android app over BT, also i created a group with the scan function. 
but.. when i try to send a message it says "unable to send message" (on both smartphones)

spleenware, 2020-02-07:
Make sure both indicators on the toolbar are solid, ie. solid white bluetooth icon, and solid white 'wifi' symbol to the right. One is whether the handset has established connection to its own Ripple device, the other is if the other ripple device is reachable, and what the signal strength is.

lyrics1988123, 2020-02-08:
hm.. looks like an issue with my smartphone maybe.. cause with my "old" CAT S50 the BT connection works (classic) and with my relative new huawai mate 20 lite it does not. it can find the ripple device and  i thought it does connect to it, but the icons on the top stays the same. any idea what could cause this?

spleenware, 2020-02-09:
With Classic, make sure you do the BT pairing first, then do the CHOOSE DEVICE in the preferences screen. Also, as each device will have the same name ('Ripple Device'), make sure only one is turned ON when you are doing the pairing/preferences.

lyrics1988123, 2020-02-09:
thank you! that was the issue, it works now :D

spleenware, 2020-02-09:
> thank you! that was the issue, it works now :D

AH, good to hear. I probably should prepare a 'trouble shooting' Wiki page or something.

lyrics1988123, 2020-02-10:
yeah the issue is, that all of your informations are spreaded in github and on instructables or where it was.. a bit confusing haha

but thanks for your help, was just testing it and i love it.