spleenware/ripple #16: TTGO WiFi Lora 32, v1 BLE screen

asavvin, 2020-02-28:
Hello!
I was able to install version of TTGO WiFi Lora 32, v1 BLE screen
But I don't see anything on TTGO screen. I am using ripple messager.
Not sure is screen supported by your verison.
Thank you

asavvin, 2020-02-29:
In addition 
Let say I have 1 TTGO connected to USB Android with Commander app and Provision.
I am trying to assign device to pager and when I start upload I am getting timeout.

The messaging app I was able to manage connection between 2  TTGOs

spleenware, 2020-03-01:
Pager devices are a little different. They set their own ID, and you import them into Commander app, if you want to be able to text them. To import tap on the SCAN menu in the top toolbar, in the Device Provision screen.