spleenware/ripple #21: Ripple app android

Commander-Crash, 2020-06-28:
I want to say this is great work and This is more of a modification than a issue. I have been using the Ripple messenger app for android for a few weeks and I notice that in the menu option you have Remove friend set up and Preferences. Not There is nothing wrong with having a menu However can you move the "Remove friend" option somewhere else. This is a issue because I do not have pencil fingers and I end up hitting that and will have to rescan the contact over and over again, and I can not scan a contact that im not next to. Perhaps you can add a long press option on the Contact list for removing contact. Thanks and keep up the grate work.

spleenware, 2020-06-28:
Good point. Hadn't thought of that. Sure, I'll redesign how the contact/friend is removed.

spleenware, 2020-06-29:
OK, I have published ver 1.6.1, which now pops up confirmation dialog when you select Remove Friend.

Commander-Crash, 2020-06-29:
Thanks. That works.

closed 2020-07-29 by Commander-Crash