spleenware/ripple #22: Would it be possible for Ripple to work with TAK?

fieldmapper, 2020-07-14:
Hi, with the TAK software open sourced here - https://github.com/deptofdefense/AndroidTacticalAssaultKit-CIV and the Software Development Kit out here http://7n7.us/civtak4sdk

A pretty ok app on what the software is here; https://www.youtube.com/watch?v=WXuQERL_e8M

spleenware, 2020-07-15:
Integrations are hard, and I simply wouldn't have the time to take on something like integrating with this app.

spleenware, 2020-09-28:
> Hi, with the TAK software open sourced here - https://github.com/deptofdefense/AndroidTacticalAssaultKit-CIV and the Software Development Kit out here http://7n7.us/civtak4sdk
> 
> A pretty ok app on what the software is here; https://www.youtube.com/watch?v=WXuQERL_e8M
> 
> Thank you very much.

Hi,

I am close to releasing a new Android app called Ripple Tactical, which will be a very simple alternative to ATAK. There is a draft user guide here in the Wiki:  https://github.com/spleenware/ripple/wiki/Tactical-app-guide

And, a new firmware will be released for the TTGO v2.1 boards