spleenware/ripple #24: QWERTY Communicators "hanging"

grinder00, 2020-08-12:
First off, amazing project, amazing work.

I have two communicators that will send their credentials to each other, and I can get to the point where one will send messages to the other, but the receiving communicator will freeze.  After a while (30 minutes) it might become responsive once again, but it certainly doesn't facilitate realtime communication.  I'm using the exact  here https://www.instructables.com/id/LoRa-QWERTY-Pager/ .  Of note, I'm replacing one of the timer chips due to a loose connection on the board.  Are the timing chips completely necessary?  Could a malfunction in one of them cause this problem?  Thanks!

spleenware, 2020-08-13:
what does the 'freeze' entail? Is the user interface still responsive, ie. you can still navigate menus etc? Is it completely locked up? Or, is the LoRa radio part 'frozen', ie. no data can be sent for 30 minutes?

spleenware, 2020-08-13:
Yes, the RTC clock modules are important. If they are not working properly, other devices may end up ignoring all data sent from it.

grinder00, 2020-08-24:
A new RTC clock module fixed the problem!!!

Couple quick questions:
1. Is the QWERTY device compatible with the V4 repeater firmware?
2. Does the TTGO board require an SD card or RTC clock when it is used in OTG Android messenger mode?
3. Is there any other way to program the repeater node other than with the Android Commander app?

Thanks!

spleenware, 2020-08-25:
AH, good to hear.
answers:
1. Yes.
2. No
3. No.