spleenware/ripple #26: Ttgo t-beam support

nlantz79, 2021-01-04:
Hey guys,  tried out the bin for ttgo lora32 on a ttgo t-beam.  Booted to the screen but couldn't connect via bt. Anyway for support of this board?

spleenware, 2021-01-05:
I haven't tested on the T-Beam, but it might be pin compatible with the other TTGO ESP32 boards. Did you flash this firmware file?   Ripple-Bluetooth.TTGOV2.bin

nlantz79, 2021-01-05:
actually using RippleTactical.TTGO.bin. Just tried Ripple-Bluetooth.TTGOV2.bin but it only boot loops.

While typing this I gave tactical another shot, it loads, shows powered by Ripple then flashes the status screen for a few seconds before going black, not sure if its a timeout or how to reactivate the display. 
On the app, it makes it to the Choose device screen where it hangs once you select ripple tactical as the Bluetooth device. Just discovered if you reset the t-beam by long-pressing the power button it will make the connection and pair. I'll test this out with a pair of t-beams tomorrow. 
Im new to these lora32 devices and assumed this is just a ttgo esp32 board with an included GPS module.  Have a pair of esp32 boards in route from china but figured I would give these a shot. 
Im trying to find an offline option to locate members of our side by side and atv group to keep tabs if anyone gets lost via location on map and text communication. This is looking more promising than meshtastic atm.

spleenware, 2021-01-05:
I have a couple T-Beam modules here, but never got the GPS to work on them. I'll have a go at flashing one of them with the Tactical firmware, and see if I can diagnose any issues. Tactical doesn't need GPS on the ESP32 module itself, getting location data from the paired Android instead.

spleenware, 2021-01-05:
Well, I tried compiling an experimental build for T-Beam, and flashed to one of my T-Beams, but the Bluetooth pairing doesn't work. It detects the device, but the pairing doesn't work. 
I got my T-Beams modules as samples, but like I said, I've never had any luck with them. Might have just gotten some bad ones, who knows, but either way I can't produce any targeted builds for these.

nlantz79, 2021-01-05:
yeah, it seems to be all over the place, I got both phone and a tablet connected/paired to separate units with the default  RippleTactical.TTGO.bin  had to wait a couple of seconds before I selected the device. or had to pair the device via the phone's Bluetooth settings first but had trouble making the devices see each other. ill give that another shot tomorrow or slate them back for meshtastic and wait for the other boards to arrive from china. 
Thanks for giving it a shot!

megabyte2010, 2021-03-24:
T-beam 1.1 needs the axp library to power all components. And if they was flashed with meshtastic, the gps stop working normally. Needs to be reseted to go back to start giving nmea sentences again.

closed 2023-05-09 by spleenware