spleenware/ripple #29: Can Heltec Esp32 Lora board be used as a tracker?

SuperSeadog, 2021-02-09:
If yes, which version of the firmware should i use?

Also, since the frequency is configurable, how do i adjust the frequency. 

Thanks in advance.

spleenware, 2021-02-09:
I have an experimental firmware version for the Heltec, actually. I'll clean it up and upload it soon so you can try it out.
It has a slight caveat, in that the serial tx/rx pins are shared with the USB Uart, so you have to boot it up in two different modes, 1) holding down the PRG button will put it in 'config' mode, so you can attach USB to Android running Ripple Commander, 2) normal power up, will try to read the GPS receiver over serial.
You set the frequency for the whole 'Group' in Ripple Commander (the pencil icon to the right of the group setting, eg. '0'). Then when you do the 'Program' steps with each node you define, the frequency setting gets sent (and saved) on the device(s)

spleenware, 2021-02-09:
OK, I have a beta firmware version uploaded here: https://github.com/spleenware/ripple/blob/master/README.md#gps-tracker-node---433-to-915mhz-configurable

SuperSeadog, 2021-02-09:
I only went as far as flashing the firmware. The LED doesn't turn on anymore. Is that normal? 

Have not connected the GPS, or tried to program with ripple commander.

OTG cable can just be any cable with the correct end connectors? 

BTW, if I uninstall Ripple Commander on my phone, can I reinstall on a different device?

spleenware, 2021-02-10:
For connecting the GPS, you can wire one of the 'Vext' pins to the GPS VCC wire, and then you can take advantage of the 'Peripheral Enable' checkboxes in Commander app. ie. some low power modes can be configured to power down the GPS while sleeping. The RX/TX (just next to the 'Vext' pins) are wired to the TX/RX (crossed over) of the GPS. 
OTG cables are different to normal USB cables. You can get adapters that clip onto the large 'A' USB end, which converts it into an OTG cable.
Ripple Commander now has 'Backup Data' and 'Restore Data' menus so that you can move everything over to another Android.

SuperSeadog, 2021-02-10:
Sorry, I meant the LCD display doesn't turn on anymore, just stays dark after flashing. Is that normal? 

"The RX/TX (just next to the 'Vext' pins) are wired to the TX/RX (crossed over) of the GPS." Do you mean connect RX (ESP board) to TX (GPS)? 

Thanks!

spleenware, 2021-02-10:
Yes, there's no OLED display defined for the GPS tracker (at present) so will stay off.
Yes, RX(ESP) -> TX(GPS), TX(ESP) -> RX(GPS)

SuperSeadog, 2021-02-10:
When programming the board with OTG cable, the ESP board would need to be self powered, such as with a battery pack? It won't be able to draw power from the Android device?

spleenware, 2021-02-10:
Oh, also when you do the 'Program' step in Commander, you will need to do the power up of the Heltec like so:
* open Commander app, using the 'Device Provision' launcher icon.
* disconnect the GPS receiver from the Heltec board. (the BN-180, for instance, just has a small 4-pin plug, which you can remove)
* connect USB-OTG cable, then *quickly* hold down the PRG button for about 2 seconds (is slightly above where the USB connector is, not to be confused with the RST (reset) button!). You need to press the button within 2 seconds of powering the board, then release it AFTER 2 seconds.

spleenware, 2021-02-10:
Ah, about the Heltec board's power. For the 'Program' step, you also need to Disconnect any Lipo battery first.
The board must be powered by just the USB-OTG cable

SuperSeadog, 2021-02-10:
I have setup the gps tracker. Anyway to test whether it's working? 

Also, how do i setup the messenger device?

I flashed the second board with Ripple-USB-Heltec_v2 and connect it to Android with OTG.

I am getting an error message when I open the Device Provision app:

Device ID 1 already exists.
Delete this first if you want to use this ID. 

What should I do next?

SuperSeadog, 2021-02-10:
Also on Ripple Commander, it only shows the gps tracker node, and when I go into the pencil icon, it will display RSSI: timed out. 

Also, the tracking options are unselected.

spleenware, 2021-02-10:
You need to launch the other icon, 'Ripple Commander' (not 'Device Provision').
The app has two main activities:
'Device Provision' -> this is for setting up the network
'Ripple Commander' -> this is for monitoring the network (you also connect the 'Home' radio here)

SuperSeadog, 2021-02-10:
How do i connect the 'home' radio in commander? By entering the "exchange key" (where is this stored)?

SuperSeadog, 2021-02-10:
Also set the frequency of the gps tracker to 915 using Device Provision. That's ok?

Thanks!

spleenware, 2021-02-10:
From 'Device Provision' , there is a 'Preferences' menu under the top-right '...' menu.
Make sure the 'USB On-The-Go' option is selected here.
Exit this part of the app, then go back into the main 'RIpple Commander' activity, then connect the Home radio with the USB-OTG cable.
If it connects properly, you should see the little Bluetooth-looking icon change to a solid white in the top toolbar (it's initially grayed out)

SuperSeadog, 2021-02-10:
It's working now. I can get the RSSI and battery readout, but not the location/gps yet.

I took it outside for a little bit, but no luck. 

It should work with any gps module? Simply connect to the pins labeled as TX and RX on the board?

Thanks much!

SuperSeadog, 2021-02-10:
I'm using this gps: ATGM336H GPS Module

https://www.tinytronics.nl/shop/en/communication/gps/atgm336h-gps-module

SuperSeadog, 2021-02-11:
Update: everything works now.

When I installed a lipo battery, the leds do not light up, and when i power is through the usb port, the leds will light. Is this normal?

When tracking is it possible to draw a line for the path taken by the tracked object?

SuperSeadog, 2021-02-11:
Can i have 2 Android devices (and ripple commander) with home radio on the same network and tracking the same tracker node?

spleenware, 2021-02-11:
I checke the specs of that GPS, and looks like it needs 5V to power it. Try soldering the GPS's VCC line to the 5V pin just next to the GND pin, rather than the Vext pin. I think the Vext pins only supply 3.3V

spleenware, 2021-02-11:
No, currently can only have one Home node, and Commander app/device.

spleenware, 2021-02-11:
BTW, use the 'label' option in Github issue tracker to submit a feature request, like the tracker 'path' you mentioned.
I've just done this as an example:  https://github.com/spleenware/ripple/issues/31

For your other question, about multiple Commander Androids being able to track a GPS tracker node, that could be a feature request (ie. label = 'enhancement')

The issue tracker is a nice way of doing this, so potentially others can view or even add to the discussion. Plus, if some resolution does eventuate, I will add to the issue conversation on how the new feature works.

SuperSeadog, 2021-02-11:
Ok, thanks for your suggestion. Everything works great now. I'll add some feature requests and enhancements for consideration/discussion shortly.

Thanks for all the help!

closed 2023-05-09 by spleenware