spleenware/ripple #31: Show GPS tracker 'trails' in Commander app

spleenware, 2021-02-11:
Provide a way to enable the capturing of GPS tracker node trails, over time, and display in the Map tab.
Quesrions:
* how is this to be stored, and for how long?
* what if the Commander app isn't running? ie. how can the data be logged?

SuperSeadog, 2021-02-23:
Other Features for Consideration:

1）Support for multiple home node in a network (e.g. to permit multiple Commander devices to monitor tracker nodes)

2) Enable someone out of network range to monitor a network, such as using a home node to transmit data over a data network (e.g. cellular/mobile) to a remote, receiving 'home' node/Commander device.

spleenware, 2021-02-23:
For 1) I have something a bit different in mind, but haven't really decided yet. Was thinking that Ripple Tactical is more suited for multiple users tracking a GPS tracker node, rather than supporting multiple 'home' nodes. Currently, Ripple Tactical supports tracking multiple nodes, but they are usually other team members. It could easily track the dedicated GPS nodes, but I'd have to change the code to prevent the messaging actions, ie. you can't 'chat' with the GPS tracker nodes.

spleenware, 2021-02-23:
For 2) I've had experimental MQTT code for quite a while now, but still haven't crystalised my thinking on it yet. But, the basic idea is for flashing the Home node, on one of the ESP32 boards, with this 'gateway' firmware that basically forwards packets to/from an MQTT server over WiFi. So, that would open up the ability to have remote clients, instead of using the Commander app. It would require you to create the console/interface though, ie. which formats the correct messages and pushes them to MQTT, and parses received messages, etc.

SuperSeadog, 2021-02-27:
cool, thanks.

Can't wait to try.

spleenware, 2021-03-13:
OK, I have release new version of the Tactical app and the Commander app, which now supports configuring the GPS tracker devices with 'Broadcast to team'. See here: https://github.com/spleenware/ripple/wiki/Commander-app-user-guide#configuring-gps-trackers

jobionekabnoi, 2022-01-18:
> For 2) I've had experimental MQTT code for quite a while now, but still haven't crystalised my thinking on it yet. But, the basic idea is for flashing the Home node, on one of the ESP32 boards, with [...]

Has the MQTT code been integerated?