spleenware/ripple #32: Mailbox setup.

Commander-Crash, 2021-02-27:
So I ran into a issue setting up a repeater with mailbox support. all is well with the repeater functions However I am unable to set up mail boxes. The set of pictures shows the errors. I did set up mail boxes on the client side of the devices via ripple messenger app. And ripple commander I programed the ttgo/ Heltec device for repeater. What is not shown is the qr codes for node setup on ripple messenger.

One more note I did look at it in SDR play and it seems like the signal are 2 different modulations. One is more spread spectrum and the other is more burst or OOK like.

My settings are:
MHz 925
group Id:13
ID: 3(Repeater)

Hardware:

3 Devices
Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
[image]
[image]
[image]
[image]
[image]
[image]
[image]

spleenware, 2021-02-28:
When the RSSI shows 'Timed Out' then the repeater currently can't be reached.
I can see in the later screenshots that the RSSI is returning a value, so looks like the repeater is reachable in those instances. But, the value is VERY low (-157). Is the antenna connected properly?
Also, the 'Create mailbox' operation sometimes needs multiple attempts before you get the 'OK' response

Commander-Crash, 2021-02-28:
It only shows -157 when i reset the usb back to the OTG cable. It will not display RSSI unless i reseat it. I tested the repeater and it seems to work however SDRPlay shows diffrent modulations when i try and send a msg with the commander app.

Commander-Crash, 2021-03-01:
I notice ripple command will not send messages but ripple messenger will without issue. Is the Command app able to send messages to clients? or is not compatible? Thanks for the reply.

spleenware, 2021-03-01:
Yes, Commander can also send messages. Have you scanned in the QR code of the other client into Commander? And, also you need to scan the QR code for the 'Home' node in the other client's app. 
(or you can use the special 'Exchange' chat room)

spleenware, 2021-03-01:
Also, when using a custom LoRa frequency, the various sensor/repeater/tracker nodes you configure (with the 'Program' button) will get the frequency setting sent to them, so those should be automatic. HOWEVER, if you change the frequency (for the Group) you then have to do the 'Program' steps again.

Also, any Messenger of QWERTY pager devices must also setup the same frequency manually.

Commander-Crash, 2021-09-22:
Sry my account got hacked and I forgot that I had this posted. This is still a no go. I have tried many different ttgo boards with same results. Can you recommend a board that this will work on?

Commander-Crash, 2021-09-22:
Can you closed this for i can not. I don't want to double post

closed 2021-09-22 by spleenware