spleenware/ripple #34: Please add support to cubecell gps and t-beam 1.1

megabyte2010, 2021-03-24:


spleenware, 2021-03-24:
I have tried porting Ripple to the CubeCell already, and unfortunately Heltec have done a real hatchet job with Arduino. It isn't really 'compatible', and breaks a number of 3rd party libraries, like RadioLib. There are numerous discussions on Heltec's own community boards, and on RadioLib's github. Hopefully Heltec will overhaul their Arduino support in a new version, because the boards look very promising. (I even bought 2 of the ones with the built in GPS)

I was actually sent 2 free T-Beam devices well over a year ago, and neither of them worked. Wouldn't even boot up properly. Probably just really bad luck, as I've seen projects like Meshtastic that have had success with them. If anyone wants to send me some ones that work I'm happy to add support for them.

SuperSeadog, 2021-05-01:
I have a t-beam. Where should i ship it?

spleenware, 2021-05-01:
Hi SuperSeadog. That's very kind of you. Can you email me at admin@spleenware.com, and I'll give you the details there?

closed 2023-05-09 by spleenware