spleenware/ripple #36: TTGO GPS tracker support

Gavsv8, 2021-03-28:
Hi, Love the project. 
I have the TTGO V2 1.6 versions (https://a.aliexpress.com/_mrWdNrt) and have been trying the various firmware targets for messenger and  tactical and can't really get what I need which I believe is the GPS tracker option. 
I want to use one as a base unit connected to an Android phone and then track the rest which will be on UAVs' with L80- M39 GPS units connected to TX1/RX1. Am I correct in thinking I can't do this until the GPS tracker is available for the TTGO? Is there a plan to add this please?

spleenware, 2021-03-28:
I have compiled a test version for you, and uploaded it here. File name is: RippleTrackerV5-TTGOV2.bin

The only caveat is that to configure the device over USB, with the Commander app, you will need to wire up a push button onto PIN 34, with a 10K pull-up resistor. See this article for an example of wiring up a push button (it has 3, you just need the one on pin 34):
   https://www.hackster.io/scottpowell69/lora-tracker-pager-device-36bdb3

Have a read of this issue, which explains why there is the need for the push button, and the configuration:
   https://github.com/spleenware/ripple/issues/29

Gavsv8, 2021-03-28:
Great thanks I'll try it now

Gavsv8, 2021-03-28:
Hi, The firmware flashed okay No oled display though. 
I could not connect until I connected the 3 inputs to the 10K pull up resistors connected to 3.3V.
I am not able to connect if I ground pin 34 on power up and get TimeoutExeption:null. 
If the 3 button inputs are only connected to the the 10K pull up resistors I can connect and the program button is available and seems to program ok. 
Once power cycled I cannot seem to connect in the commander app and it mentioned unsupported firmware version.

No oled display in any mode.

Gavsv8, 2021-03-28:
I configured two TTGO with the special tracker firmware and also flashed a third with the v3 ripple ttgo and configured this as a repeater to see if it would work. In commander repeater is connected but nothing on map apart from the andriod location.  I can't tell if gps is working on the other units as there is no oled. I'm in Perth working nightshift so doing some weird hours at the moment. Can try some more stuff tomorrow night, Cheers.

spleenware, 2021-03-29:
Ah, I got the button wiring around the wrong way.
To configure the tracker, ie. to have USB OTG work, just wire PIN 34 to a
10K resistor and the resistor to 3.3V pin. (ie. pull it 'high').
Then once configured, remove that wiring, and connect PIN 34 to a GND pin
(ie. pull low). From this point on, the RX/TX pins should be wired to the
GPS receiver.

spleenware, 2021-03-29:
The 'Home' TTGO device, that you connect to Commander app to monitor the
trackers, should be flashed with one of these two firmwares:

RippleV3-USB-TTGOV2.bin  -OR-  RippleV3-Bluetooth-TTGOV2.bin

ie, the 'Pager/Messenger' firmware.

Gavsv8, 2021-03-29:
Yep thanks that's all good now.
I can see the remote units RSSI etc and they respond to tracking mode changes. 
I'm not getting any location details just 0.0, 0.0 (1,0,0) but I'm thinking it maybe an issue with the Quectel L80-M39 GPS units and will try a BN180/220 instead.

Gavsv8, 2021-04-02:
Hi, 
It's all working fine now I have changed the GPS to a ublox M8N (matek m8Q). Thanks for the work on the new firmware.

spleenware, 2021-04-02:
Ah, cool. Glad it's working now.

closed 2021-04-03 by Gavsv8