spleenware/ripple #37: Flash to Feather M0 LoRa unsuccessful

JH2OSolutions, 2021-04-04:
Attempting to flash Version 3 Firmware for Adafruit Feather M0 (board is Feather M0 LoRa 900mhz version) from Ubuntu 20.04 LTS.  I was able to edit the text to find binaries and to use the correct port (as displayed in IDE).  When I run command, I get the following:

Set binary mode
Send auto-baud
Set binary mode
No device found on [Port]

Port listed in IDE when device is connected is as follows:

"/dev/ttyACM0 (Adafruit Feather M0)"

I have tried it with and without the quotes, with and without the portion in the parenthesis.  It does not work unless I edit it as follows (with no quotes and not parenthesis):

--port=/dev/ttyACM0

Except I get:

No device found on /dev/ttyACM0

What am I doing wrong?

Thanks,

spleenware, 2021-04-04:
Have you been able to flash a basic Arduino sketch onto the feather? Try
loading up any of the example sketches, like the Led blink one, and run
from IDE. If you get an error, there might be better diagnostics to trouble
shoot. But if it does work, capture the command line invocation that the
IDE generates by ticking the 'Verbose' option under Arduino preferences for
the 'Run/upload'

It should look like the command line commands I have pasted into the Readme
on this GitHub project. Substitute the .bin file for the Ripple feather
bin, and it should be good

JH2OSolutions, 2021-04-04:
Interesting.  I am learning a lot so thanks for that.  I was able to upload the Blink sketch and it seemed to work.  Here is what IDE spit back out at me with verbose option selected:

/home/user/snap/arduino/50/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3/bossac -i -d --port=ttyACM0 -U true -i -e -w -v /tmp/arduino_build_759043/Blink.ino.bin -R 

Here is what I was trying before:

/home/user/snap/arduino/50/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3/bossac -i -d --port=ttyACM0 -U true -i -e -w -v RippleV3-USB-feather.bin -R 
I still am having the same error.  Here is what I get when I attempt to run the command:

Set binary mode
Send auto-baud
Set binary mode
No device found on ttyACM0

Thanks for your assistance.

spleenware, 2021-04-04:
Hmm, not sure. Sometimes with the feather if port not found, you can press
the reset button on the feather twice, quickly, and it will out the board
into upload mode. The led stays on, but fades in out to show the upload
mode is active

JH2OSolutions, 2021-04-04:
Ok.  That made a difference.  Here is what I got back:

/home/user/snap/arduino/50/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3/bossac -i -d --port=ttyACM0 -U true -i -e -w -v RippleV3-USB-feather.bin -R 
Set binary mode
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
version()=v2.0 [Arduino:XYZ] Mar  5 2016 17:46:52
chipId=0x10010005
Connected at 921600 baud
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
Atmel SMART device 0x10010005 found
write(addr=0x20004000,size=0x34)
writeWord(addr=0x20004030,value=0x10)
writeWord(addr=0x20004020,value=0x20008000)
Device       : ATSAMD21G18A
readWord(addr=0)=0x20007ffc
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
Chip ID      : 10010005
version()=v2.0 [Arduino:XYZ] Mar  5 2016 17:46:52
Version      : v2.0 [Arduino:XYZ] Mar  5 2016 17:46:52
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
none
readWord(addr=0x41004018)=0
Security     : false
Boot Flash   : true
readWord(addr=0x40000834)=0x7000a
BOD          : true
readWord(addr=0x40000834)=0x7000a
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
chipErase(addr=0x2000)
done in 0.842 seconds

No such file or directory

I am troubled by that last line.  Is that normal?

spleenware, 2021-04-04:
You might need to provide full path to the Ripple bin file

JH2OSolutions, 2021-04-04:
Ok.  The full path worked.  In retrospect, I should have realized why it wasn't working before.  I thank you for your assistance.  I will experiment with it now and see if I can't figure it out.