spleenware/ripple #40: Add button to read messages on device.

Etayson, 2021-08-12:
Hello. First, I want to thank you for this project.
Is it possible to make a new version of the RippleV5-Bluetooth-TTGOV2 binary that will support the button (attach it to a free pin) for reading messages on TTGOV2 device? Those to be able to use it as a pager, because now the messages on the screen disappear after a certain time. So the button is needed to display one by one all the messages that have accumulated in the device's buffer and clear buffer after reading all message on the device. Also this button can display status screen when there no messages in buffer.

spleenware, 2021-08-12:
Well, as it happens, I am working on something that does include what you've described. The idea is to have the Ripple radio still as a companion to the Messenger app, but be 'detachable' so to speak, and usable on its own just for receiving and reading messages. ie. with a simple UI of its own. Here's a pic of current prototype:
[image]

Etayson, 2021-08-12:
Great to hear this news!
I will wait new version of firmware and schema. 
Tell me please the address where I can send a donation to motivate you.

spleenware, 2021-08-13:
My BTC address: 1CspaTKKXZynVUviXQPrppGm45nBaAygmS

Much appreciated.

Etayson, 2021-08-13:
I sent you a gift. There is also a question whether it is possible to add сyrillic font support to the device TTGO. If i send a message in cyrillic, then it arrives on the phone messenger correctly, but it is not displayed on the device. Thanks!

spleenware, 2021-08-13:
Wow, you are a saint! Thank you so much. I will see if there is a Cyrillic font that I can compile into the device code.
Also, email me at: admin@spleenware.com
I'll make sure I send you all the early releases, etc.

closed 2021-08-13 by Etayson