spleenware/ripple #41: Mail Box Setup.

Commander-Crash, 2021-09-22:
A while back I posted this issue about mailboxes in Ripple commander. I have yet to figure out what is wrong. After setting up the repeater and programming it with provision. I head back to ripple commander and it will say timeout or -157db. So I set this aside for a few months and found an update. I did the update and still have this issue. Is there a special board that this will work on IDK if anyone else has this issue.

Boards I have Tried:
LILYGO TTGO SX1278/SX1276 http://www.lilygo.cn/prod_view.aspx?TypeId=50003&Id=1143&FId=t3:50003:3 
ESP32 LoRa 32 (V2) https://www.amazon.com/gp/product/B076MSLFC9/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B07B95BY52/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1 #Flashes but no oled support no connection with BT
T-Beam #will not flash

All Boards Come back "null" "time out" or "-157db" If there is a board that works with mailboxes pleas post a amazon or ebay link. Thanks.

spleenware, 2021-09-22:
The second one looks like the Heltec boards (prob a clone). Have you tried this firmware with those? https://github.com/spleenware/ripple/blob/master/RippleRepeater-heltec_v2.bin

Commander-Crash, 2021-09-22:
Thanks for the quick reply. I have tried that .bin file and it will flash to the 1st board but same results.

spleenware, 2021-09-22:
Just trying to re-read your older mails to refresh my memory. I recommend trying all the defaults first, ie. 915Mhz, SF 10, group 0. See if all the defaults work, which is the most likely. Then we can maybe try to nail down why some settings aren't working. Also, note that if you change any of the config in Commander, you need to re-program the various devices (except the Home (ID:1) radio).
Which board are you using as the Home radio? And, which firmware did you select?
For the Heltec-clone, you would use:
https://github.com/spleenware/ripple/blob/master/RippleV5-Bluetooth-HeltecV2.bin
(and connect to it using Bluetooth)

spleenware, 2021-09-22:
Also, with the Home radio, there is an existing bug if you previously flashed that device in some OTHER role (eg. as sensor or repeater). It can still pickup various config from its old role, which can override the config it should be using. You can do a full flash erase, and start from scratch to get around this.

Commander-Crash, 2021-09-22:
I will do the full flash i had no ideal about the bug nor that radio home (ID:1) needs a board as well.

Commander-Crash, 2022-08-27:
This issue can be closed it was due to my hardware. Different types of lora32 devices some work well while others don't work with mailbox setup. Some ping successfully while others you can't get a response. LiliyGo TTGO V1's is all i had but now have 2 t beams and they work fine.

closed 2022-08-27 by Commander-Crash