spleenware/ripple #46: Xiao Ble or Xiao Ble Sense compatibilty

Telepresent, 2022-03-15:
Would be nice to have Xiao Ble or Xiao Ble Sense compatibilty. They use Nordic Chipset. 

Regards

spleenware, 2022-03-16:
Which particular firmware are you looking to use these in? For use with the Messenger app, or Tactical, or as the 'home' radio for Commander?

spleenware, 2022-03-18:
Also, just price-wise, I find this Heltec to be great value: https://heltec.org/project/wireless-stick-lite/
This is my go-to board now. I use this for about 5 different node types now. (Repeater, Qwerty, etc)
The Xiao Ble + some Lora module would be about the same price. (maybe a bit more, actually)

I don't mind adding support for this. If I could get a sample unit sent to me, that would help a lot :-)

closed 2023-05-09 by spleenware