spleenware/ripple #52: Which code to use to flash

17dshaikh, 2022-10-09:
Hi I am wishing to create the gps lora tracker but I am confused I don't know which code to use to flash the gps lora boardand what code to flash the board for the commander app please help.
I am using the adafruit lora board 433mhz
Thanks

spleenware, 2022-10-10:
I'm assuming you're using two Feathers, one for the tracker, one for the 'home' radio (connects to Android and the Commander app).
For the 'home' radio, flash this: https://github.com/spleenware/ripple/blob/master/RippleV5-USB-feather.bin
For the tracker, flash this: https://github.com/spleenware/ripple/blob/master/RippleTrackerV5-Feather.bin

17dshaikh, 2022-10-10:
Should I use arduino ide to flash it or something else?

17dshaikh, 2022-10-10:
Sorry pressed wrong button

17dshaikh, 2022-10-10:
As the guide mentions a flasher but also ardunio ide

closed 2022-10-10 by 17dshaikh

spleenware, 2022-10-10:
The EspHome flasher tool is by far the easiest way

17dshaikh, 2022-10-10:
Do you have a link for the download of that program

spleenware, 2022-10-10:
https://github.com/esphome/esphome-flasher/releases