spleenware/ripple #56: ERROR: id incompatible [TTGO v1.6]

641i130, 2022-10-20:
I've flashed it using this command:
```
esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 ~/Downloads/boot_app0.bin 0x1000 ~/Downloads/bootloader_dio_80m.bin 0x10000 RippleQwerty-TTGOV2.bin 0x8000 RippleQwerty-TTGOV2.partitions.bin 
```
I then added this to the conf.txt in the ripple directory:
```
GR0
ID14
SF11
FQ915
Tom
TZ600 
```
and for my other one:
```
GR0
ID13
SF11
FQ915
Boots
TZ600 
```
I then turn them on, use the password 1234 for example and I get an `ERROR: id incompatible` whenever trying to send data on Exchange.
I believe the firmware could be broken...
I would like to help fix this but there is no source code. 

I'm building this (minus the RTC and buzzer):
[image]

641i130, 2022-10-20:
I've also forgot to mention that I've used `dos2unix` thinking the file format might matter (since I'm on UNIX). I noticed you changed it so dos type files would work in the comment section. This made no change to my error.

spleenware, 2022-10-21:
You forgot to put 'NM' before the names. eg. NMTom

641i130, 2022-10-21:
🤦I can't believe it was this lol

Will you ever open source the code?

spleenware, 2022-10-21:
There hasn't been any compelling reason to do so yet.

641i130, 2022-10-21:
Personally, I'd like to see what the community contributes, plus I'd love to see the inner workings since I'm currently enrolled in university. I've been studying embedded systems and love LORA! Clearly, this project is one of the top ones for LORA communication. Although, I can understand why you'd want to keep the code to yourself (if you're making a profit off of it).
I just found it odd that this project was sort of in a half-open-source state. You have everything else public, why not just put the code so people can help contribute/modify? Many other projects benefit from this system.
If you don't want to manage the code, having it out there lets others fork it and begin a branch of a new project too.
It's up to you though.
Anyways, thanks for this amazing project! And the support you give us. I do hope you'll give releasing the code some thought. I'd love to help you clean up the repository (it is quite unorganized!!!).

closed 2022-11-07 by spleenware