spleenware/ripple #58: Tbeam as home unit. Is it possible?

Postsitter, 2022-12-10:
I downloaded the app and flashed the RippleV5-Bluetooth-TBeam.bin on a T-beam as I believed it is for using a t-beam as a home unit. Unfortunately I cannot get it to work. Do I have to buy a TTGO LoRa32 V2, or Heltec V2 for the home unit?

spleenware, 2022-12-14:
You should be able to use the TBeam as home unit, but you might have a later TBeam which has the SX1262 LoRa module. Do you know if that's so? The RippleV5-Bluetooth-TBeam.bin firmware is for the (now) older one, which is the only one I have. I could produce a new firmware bin for that one, for you to try out. Just let me know.

Postsitter, 2022-12-14:
It is a Tbeam v 1.1. According to the text on the board. How can I check
the lora chip version?

Postsitter, 2022-12-18:
It works now. Just me doing user faults. Thanks. Great system!