spleenware/ripple #61: Max devices per mesh and max hops

danilomna, 2023-03-19:
Hi

How many devices are possible on a single ripple mesh network?

How many hops are allowed? Max hops number?

Thanks in advance.

spleenware, 2023-03-22:
There are a maximum of 254 physical devices per mesh.
Max hops is difficult to answer. There is a 'practical' maximum based on the latency introduced with each hop, which depends on spreading factor. I don't have the time-on-air calculations on me, but you'd have to add those, but there are small delays of around 300-600 milliseconds (for SF=10), for example. 150-300 millis (for SF=9, my favourite).
A total round trip will experience a timeout (in the Android apps), for a message send, at 4sec on 1st attempt, 6sec on 2nd, 9sec on 3rd. So, technically, if the round trip can be done in just under 9sec, it could have done 15+15 hops, for example.

danilomna, 2023-03-22:
Ok, understood, thanks for explain.

Is this 254 devices limit due to RadioHead Mesh library?

danilomna, 2023-03-23:
Understood, great.
How hard would be to make it work with 512 or 1024 devices in a single mesh work? Is this possible to customize?

closed 2023-05-09 by spleenware