spleenware/ripple #63: A few questions

LuxStatera, 2023-05-06:
I have 3 ESP32 TTGO v2 boards. I first tried the SMS app with 2 of them and was able to exchange messages. I tried to make the 3rd into a mailbox/repeater but it didn't work. I read that I needed to make 1 of them a repeater and set it up as "Home". So now I have 1 as a messenger, 1 as a Home node, and the last one as a mailbox/repeater with solar power/batteries. Is this the correct setup? The "Home" node works as the admin control but does it also act as a repeater? 

[image]
After I scan the QR code for the mailbox/repeater I get a notification to power cycle the Ripple device to update changes. Once I reconnect I get this error message and the repeater doesn't show up on my Ripple Messenger list. I can see it on Ripple Commander and it shows it is on but I am not sure if it is truly working since I can't add it to my list.

In the Ripple Commander app there is an option to create a group chat but it won't let me. I get this error message each time:
[image]

Unfortunately I can't seem to get it all to work properly. Is there something I am missing?

spleenware, 2023-05-06:
When you did the Device Provision, and added the repeater node (ie. assigning the ID/name), did you enable the 'Post Office' feature. In the device setup screen (ie. where you enter ID/name) there is a '...' button. When you tap that there are extra config options. You need to check the 'Enable' checkbox and enter a number for the 'max messages per recipient'.
(You can re-configure the repeater at any time with a USB-OTG cable. You need to click on the 'Program' button once you've made changes)

Once the Post Office is enabled, you should see 'Create Mailbox' and 'Remove mailbox' buttons in the device info screen for the repeater (ie. back in the main Commander screens, not Device Provision). If you don't see those buttons, then the Post Office feature hasn't been enabled.

You also need to scan each Messenger user's QR code into Commander, so that it gets a copy of their ID/Name/Public-key. Then, you send the user's details to the repeater with the 'Create mailbox' button.

In Messenger, when you scan the repeater's QR code, it gets stored in Preferences only, and doesn't show in the main list of contacts. So, it's a two-way key exchange: user needs repeater's id/pub-key, repeater needs user's id/pub-key.

Also, the 'pager' firmware can be used with both Messenger and Commander (as Home node), but obviously not at the same time. So, one of them can temporarily be Home node 1, while configuring, then you can test on Messenger as another user with the same TTGO. (each time it connects, the settings and contacts get wiped).

LuxStatera, 2023-05-06:
When adding the repeater I enabled it as a Post Office and setup 10 for max messages. I programmed it once I finished with the settings. In the Commander app I clicked "create mailbox". I also added all the QR codes in the sections that allowed me to. In the Ripple Commander app I see 1 pager and 1 repeater but I do not see the Home node. In the Device Provision app I can see all 3. In the Ripple Messenger app I only see Home and it is setup to be used with the pager. How do I do the 2 way key exchange? I tried to create a Home node with pager firmware but then it didn't give me any options in the Device Provision app so I flashed the repeater firmware to allow me to set it up as a Home node

LuxStatera, 2023-05-06:
[image]

All 3 are ON and online but the pager device always shows up as red

spleenware, 2023-05-07:
The Home node is 'automatic', ie. you shouldn't try to create it (in Device Provision). And only use a radio flashed with one of the pager firmwares. The pager radios don't have any fixed ID, they just become the ID of whichever Android app you use; so if you connect to it with Commander, it gets ID 1 automatically; if from Messenger it gets whatever ID you have in Setup.

In Messenger, looks like you have setup ID 7, and that Home is listed, so you must have scanned the Home node's QR code. And obviously you have scanned Node 7's QR code into the Commander app (so the key 'exchange' is done). If it still not responding, there must be some mismatch in the radio config (Frequency and Spreading Factor). I
n Messenger this is in the Preferences screen. In Commander you set it for the 'group'. (you can manage multiple meshes, each with different Group ID) For simplicity, just stay on Group ID zero. In Device Provision, at the top, to the right of 'Group ID: 0' click the pencil icon. Here you set the Freq/SF for the group. Make sure these match what is in Messenger, or vice versa.

spleenware, 2023-05-07:
Oh, also I just noticed that in the screen shot, the node 7 is slightly greyed out. This is an indication that this node hasn't had its public key sent to the Home radio. If you tap on the node 7 item and go into the conversation screen the key will be sent (over Bluetooth or USB-OTG) to the Home radio. And sending/receiving messages should work.
If you go back to the main screen, you should see that node 7 isn't greyed out now. Clicking on the icon to the right (to get status) should return green.

LuxStatera, 2023-05-07:
[image]
It works! Thank you so much for all the help! I am waiting on my 4th LoRa ESP32 to arrive so I can make another pager. After that I will deploy the solar mailbox out in the wild :)

LuxStatera, 2023-05-07:
I was also wondering about the setup. If I deploy the Solar Post Office out in the wild and use the the 2 pagers while I am mobile then would the Home node constantly stay ON at my house? Or does it need to stay with a repeater or mobile with me? If it stays at the house then could it act as a repeater so I can put an antenna on my roof?

The ideal setup being a node at home that acts as a repeater and post office, solar powered repeaters out in the wild, mobile devices for on the go, and something at home that is always ON to accept messages incase of an emergency and/or if my mobile device is not ON. It would be handy to have a "base station" that is a post office, repeater, and messenger. Maybe the Home node could do this?

spleenware, 2023-05-07:
Typically, you only use the Home node intermittently, just to check on the status of repeaters, or to configure stuff. Not usually an always-on thing, unless you have sensors or GPS trackers deployed that are configured to generate alerts.
And, like I said, the home node doesn't have to use a dedicated TTGO radio. I have the one pager device I alternate between using with Commander and Messenger, for instance.
The Post Office node is also a repeater, like all the nodes. (unless you specify Mesh Role: 'End Node' in the config) Ideally you put the single Post Office repeater up somewhere high, like up a mountain or hill, that is maximally reachable. So, when anyone is sending a message, it has the most likelihood of at least reaching the Post Office (where the messages sit in store-and-forward queues). You could also put the Post Office node on your roof at your home, but with line of sight to another plain repeater up on a hill.
Also, I highly recommend either the Heltec V2/V3 or Heltec StickLite V3, for repeater builds. The repeater firmware for these can take advantage of special low power modes, which put the repeater into deep sleep with wakeup on hearing LoRa activity. These require much less solar to keep them topped up.

LuxStatera, 2023-05-07:
Okay I understand about the Home node now. I used another phone to setup another pager with the same ESP32 I used for the Home node. I am able to send messages between the two devices. How do I utilize the Post Office? If I try to send a message to a device that is offline then I get an error message. Also in the Commander app there is a folder icon where I can create group chats but when I try to create one I get an error. I have the solar repeater enabled as a post office but I don't think it is working

spleenware, 2023-05-07:
Both Messenger users need to have scanned the QR code for the Post Office (in Preferences screen), and both users need to have had the Create Mailbox completed. And, obviously each Messenger user needs to scan each other's QR codes to get into each other's contact list.
Also, which firmware .bin did you used to flash the Post Office node? The TTGO repeater firmware might be a tad old :-( I've only been working with the Heltec boards for repeaters, and the TTGO version might have been neglected. Let me check.

spleenware, 2023-05-07:
Hmm, OK. Wow, the TTGO firmware was super old :-( I have updated it for you to try. I haven't tried this combo (repeater + TTGO) for a looong time, so hopefully it works. Firmware here:
   https://github.com/spleenware/ripple/blob/master/RippleRepeater-TTGOV2.bin
I've also updated the Bluetooth pager firmware:
   https://github.com/spleenware/ripple/blob/master/RippleV6.2-Bluetooth-TTGO.bin

LuxStatera, 2023-05-07:
I'll try the new firmware and check back momentarily

LuxStatera, 2023-05-07:
I tried the new firmware and reflashed everything. I deleted all the old settings and tried again by following all the steps we discussed. I can see the devices online and can send messages but when a pager is offline then I can't sent messages to it. I get this error when trying to message an offline device:
[image]

Also when trying to create a group chat I get this error message: 
[image]

spleenware, 2023-05-07:
When you had the old repeater firmware, I'm guessing the 'Create Mailbox' steps would have failed(?) Did this work with this new firmware? A 'Success' popup should come up once you select a user to create mailbox for. If not, then something is wrong.

LuxStatera, 2023-05-07:
I was successful when creating a mailbox on the old firmware and new firmware

spleenware, 2023-05-07:
Can you time with a stopwatch how long from when you press the send button to when the 'max retries reached' comes up? That could give me a clue.
Also, is this between two Messenger users? Just wondering if you're trying to send a message from Commander to an (offline) Messenger user? To keep is simpler, for now, just should be trying Messenger app to Messenger app (with one being offline).
And both Messenger apps have in the Preferences screen, at the bottom Post Office section, has a 'Node: xxx (name)' where xxx is whatever your post office node ID is, showing that both have the (same) Post Office configured?

LuxStatera, 2023-05-07:
With a stopwatch it took about 20 secs. Messenger app to Messenger app. Works with both devices are online. Error message when one is offline. They both have the same exact settings besides the name and ID. Both have Solar Post Office as their post office

LuxStatera, 2023-05-07:
I'm also getting this notification quite often in the Commander app
[image]

spleenware, 2023-05-08:
Ah, the 'sync-clock' is a new convenience in the repeater firmware. Whenever it is rebooted, the clock needs syncing, and previously you had to remember to click the 'force refresh' button in Commander (in the device info screen for the repeater node). Now, the repeater asks the Home node for the current time automatically.
The pager firmware needed a tweak to make this happen. I've updated the TTGO firmware:
   https://github.com/spleenware/ripple/blob/master/RipplePager-BT-TTGO-433mhz.bin
   https://github.com/spleenware/ripple/blob/master/RipplePager-BT-TTGO-915mhz.bin

Regarding the post office, I'm going to have to setup a TTGO repeater here and try to reproduce. I haven't used the TTGO for a repeater before, so there must be something weird/odd going on with that firmware/hardware combo.

spleenware, 2023-05-08:
OK, I setup a TTGO repeater from scratch, and set up two Messenger users, etc. Worked fine. (prob not what you want to hear). So, I think it's safe to assume Commander <-> repeater comms works (sounds like it). And Messenger <-> Messenger comms works. But something is up with Messenger <-> repeater. The only thing I can think of is:
  * Commander and repeater are working on a different frequency and/or spreading factor to what they Messenger ones are.

LuxStatera, 2023-05-08:
I tried the new firmware but running into the same issues. All ESP32 show the same frequency and spread factor in their preferences. I can send messages between pagers and I can see the repeater is online in the Commander app. Unfortunately I can't tell if the repeater is working because I have them too close together to really test and also the post office function isn't working. What should I normally expect when sending a message from an online pager to an offline pager?

LuxStatera, 2023-05-08:
[image]
Also noticed this

spleenware, 2023-05-08:
OK, might be able to do some sleuthing via the DiIagnostics button. In Commander, go into the device info screen for the post office, and tap on the Android icon in top right. Should see 'Device Diagnostics' dialog pop up.
Take note of 'Packets Recv:' counter, and the 'Packets Ignored:' counter.  
Turn off the Home radio.
Then try send a message in one of the Messenger apps to an offline Messenger. 
Turn home radio back on, wait for BT connection to settle , then tap on the Android icon again to get new Diagnostics, and compare the counters.
Let me know what the before and after counters are.

spleenware, 2023-05-08:
It seems to be that the post office is not recognising the packets sent from the Messenger, and my guess is that it doesn't have the right public key for the sender, or vice versa, the sender doesn't have the right public key for the post office.
I'm really at a loss as to why this is so, with your setup. But, that's all I can think of without having the actual hardware in front of me, that the key exchange is somehow wrong or didn't go properly.

LuxStatera, 2023-05-08:
I realized what I was doing wrong. I apologize because this was a foolish mistake on my part. When creating a "mailbox" I was thinking that the Post Office should only be the mailbox but I needed to add all devices as mailboxes. Now everything works. Offline messages and group chat! Here are 2 screenshots of my post office diagnostics since getting everything to work:
[image]
[image]

LuxStatera, 2023-05-08:
[image]

spleenware, 2023-05-09:
Ah, fantastic!  The group chat is a really fun feature, which I think you might be the first to actually have setup :-) You can post multiple choice surveys and quizzes. For testing, I setup a 'Daily Quiz' chat group with my family, with Write access only to me, and I'd post a general knowledge quiz question each day. Was a fun test :-)

Get in touch via email and can discuss other tidbits and tips:  ripple_biz@protonmail.com

LuxStatera, 2023-05-09:
Thank you so much for all of your help! I love your project! I'm going to test out the surveys once my buddy gets his pager :) Thanks I'll send you an email right now

[image]

I'm going to get a waterproof enclosure and a better antenna. If that works out well then that will be standard for all of the repeaters I build :)