spleenware/ripple #70: TileDownloader sd card init fail

Ketse89, 2024-05-16:
Hi, I just got two T-Decks and tried the Ripple ultra firmware: it was super, nice work!

But when I tried the Tiledownloader and uploaded it with Arduino IDE, I get on serial monitor "SD interface init fail".

I've tried different 4GB and 32GB cards. Formatted to FAT32 with Win10 and also tried official SD Card formatter.

Ketse89, 2024-05-18:
Nvm I just got similar python script and got some tiles downloaded

closed 2024-05-18 by Ketse89

spleenware, 2024-05-21:
Hmm, is a bit worrying as the SD init code in the downloader is the same as in the Ultra firmware :-(
I have only had that error when there is no SD card inserted when it starts up. And, my second guess would have been whether it's FAT32 or not...  Not sure what else to recommend. I'd need to be able to reproduce the conditions to solve it :-(

Ketse89, 2024-05-21:
I don't think there's nothing wrong with your code, I probably messed up the bootloader section or something:

1st T-deck I tested with had Lilygo factory firmware -> Adafruit online flashtool -> 10000 address RippleQwertyGeo-TDeck-sx1262-ultra.bin -> Your firmware booted without SD card init failures, no problems.

2nd T-deck I tried using as Tiledownloader -> chose wrong board and/or wrong settings in Arduino IDE -> after googling found right settings and got tiledownloader responging to Serial Monitor -> SD card init failure.
-> After that flashed your firmware to address 10000 -> SD card init failure
-> after googling flashed Lilygo factory firmware to addr 0 -> then flashed your firmware -> worked

sisari, 2024-08-18:
> Nvm I just got similar python script and got some tiles downloaded

Can ypu please point me to the Python version of tile downloader  ?

Ketse89, 2024-08-18:
> > Nvm I just got similar python script and got some tiles downloaded
> 
> Can ypu please point me to the Python version of tile downloader ?

I don't remember the exact one, but googling "openstreetmap tile download python script github" I found this that looks similar to code I used:
https://gist.github.com/tonyrewin/9444410